ni/monodrive-documentation #170: The Semantic Camera Values need a paragraph on how to set them on objects in the Editor.

stephendb, 2020-07-10:
Add screen shot and some text on how to modify/edit/add the semantic camera value, i.e., set the custom depth stencil value.
https://monodrive.readthedocs.io/en/latest/monoDrive_home/Camera/

katiemacaulay, 2020-07-13:
@stephendb, I will need more information to add this to the documentation because I am not familiar with how to do this. Would you be able to write up this description or can you recommend someone who can help with this specific task?

BrettMonoDrive, 2021-01-08:
Add a how to add semantic labels to my doc

https://docs.google.com/document/d/1iljdCm0Qnd2K79vM236mpoWLb_wSeqpb6YmC9m_8JIs/edit?usp=sharing

citlaligm, 2021-01-11:
Brett, can we add to the this document why we want to add this values? i.e. for the semantic camera to paint the actor correctly, etc?

Also, is the link you have referenced the correct location to the documentation? 
[image]

BrettMonoDrive, 2021-01-11:
Sure I can add a 'why', also the last time I checked that is the spot that has the depth values so the link should be good. But I'll check and see if Katie changed it.

BrettMonoDrive, 2021-04-08:
The Semantic labeling has been addressed in this document: https://docs.google.com/spreadsheets/d/148UZ8KIqxBZljg8LNLUnIMIKXu5U8upATmI-fUGp2ms/edit?usp=sharing

closed 2021-04-08 by BrettMonoDrive